matfroh/sax_battery_ha #21: prepare new service to choke battery

matfroh, 2025-03-23:
As per the latest [documentation](https://sax-power.net/download/Handbuch_SAX_Home_Plus_7,7_DE.pdf), in page 47, we read the future possibility to steer the battery without any register to be activated.

It will require writing on slave ID 123
|  Protokolladresse  |  Interne Adresse  |  Bezeichnung  |  Einheit  |  Unit-ID |  Hinweis |  Datentyp |  Zugriffsart |
|---|---|---|---|---|---|---|---|
| 40349  |  348  |  Sollwert Leistung  |  Steuerungsmodus (Choking) |  % | 123 | Faktor 100 -10000 … +10000 | int16 | Read/Write |
|  40353 |  352  |  -  |  123  |  | 0 = Steuerung über Smartmeter
 1 = Sollwertsteuerung. Wird über 5 Minuten kein neuer Wert gesetzt, aktiviert sich automatisch die Steuerung über
 Smartmeter (0) wieder.  | int16 | Read |

There should be a service in a service.yaml file, a number in the number.py file and everything needed in __init__.py file to have the 2 sensors available for anyone.
It seems register 349 will need to set a percentage scaled up, minus or positive which will use the power of the battery.

matfroh, 2025-03-30:
prepared in [beta release](https://github.com/matfroh/sax_battery_ha/releases/tag/v0.3.0.1-beta)

tguett, 2025-10-07:
Hallo can we get in contact to sort out some issues that i have ?
Best regards
   Thomas

tguett, 2025-10-07:
> 👋 @tguett <https://github.com/tguett> , happy to discuss here unless you
> want to send me a private message on the HA community forum.
> <https://community.home-assistant.io/t/integration-of-a-sax-battery-over-modbus-tcp-and-automation-for-solar-charging/836369>
> I had originally created that bit of code in case the new firmware is
> released, but that will only work with HA 2025.8, as code dependencies have
> evolved.
> Check in the branches of the main code and you will see a second iteration
> of it. I had since to continue to evolve the main code and there is a
> version 3 in the building in a PR. So focus on this potential new firmware
> has shifted.
> This was all just based on the documentation above and having seen a
> leaked doc of the future firmware, I prefer on my end to wait till it is
> released and prefer to help what a contributor is pushing on a PR a better
> cleaner current code. But happy to chat!

Hallo i think i have some General understanding issue in how the sax
storage with sax smartmeter ist working. I only wanted to Setup a charging
of the storage during night hours with ceap Energy. I enabled all the
Registers via sax Tool. Now the storage hast lost tge Connection to the sax
smartmeter and is Not working at all again. So If you have any Idea to fix
this again would bei very appreciated. :-(  Best regards Thomas

matfroh, 2025-10-07:
@tguett  what your mentioning is not related to this particular issue.
Please read the doc again, either you let the smartmeter steer the battery either you let this extension steer it. You cannot with the current firmware do both. The next firmware will allow it but our code here is not ready for it.
So go what you want to do, install the latest code and enable the pilot to steer the battery and then create your automations on top!

matfroh, 2025-10-07:
@tguett i have created a new issue for this and need your input. The use case you mention was not foreseen but might be possible.

tguett, 2025-10-31:
> @tguett <https://github.com/tguett>
> Hey Thomas,
> I think you can only connect the SAX meter to the battery itself as it is
> a RS485 connection.
> If you have it already installed, you should get its data out of mats
> integration.
> You then can use the data to steer your battery as described by me in the
> discussions section.
> You have to place a change request on the SAX servicesite to change your
> system from smart meter mode to TCP mode. This is mentioned on page 45 of
> the manual.
> Ask them to change the powergradient too, as it is set to 0.27%/s. Mine is
> now at 10%/s (possible power change of ~400 W/s for quick reaction.

OK will Check. Thank you. Will let you know