matfroh/sax_battery_ha #31: energy consumption and produced need refactor

matfroh, 2025-03-26:
To match what HA is expecting, energy consumption and produced sensor need rewriting.
The value provided by the battery is an hourly consumption/production.
Total_increasing is not enough to some it over time.

Bertk, 2025-03-30:
The energy dashboard with hourly updates is also showing some bogus energy consumption/production. Currently the scan_interval is one hour (3600 seconds) which should be changed to 2 or max. 5 minutes.  

https://github.com/matfroh/sax_battery_ha/blob/80b06fa385295d2a30e00f440a054247b258ed1b/custom_components/sax_battery/__init__.py#L204-L217

matfroh, 2025-03-30:
Frequency of call is not the issue, it's the way HA expect the value. It want's a sum by calls.
I believe that a call every hour should provide the value expected, but it needs to be summed to previous value. Something like this
````
class SAXBatterySensor:
    """Base class for SAX Battery Sensors."""

    def __init__(self, battery, battery_id) -> None:
        """Initialize the sensor."""
        self.battery = battery
        self.battery_id = battery_id
        self.cumulative_value = 0  # Initialize the cumulative value

    def update_cumulative_value(self, new_value):
        """Update the cumulative value with the new hourly value."""
        if new_value is not None:  # Ensure we have a valid value
            self.cumulative_value += new_value

class SAXBatteryEnergyProducedSensor(SAXBatterySensor):
    """SAX Battery Energy Produced sensor."""

    def __init__(self, battery, battery_id) -> None:
        """Initialize the sensor."""
        super().__init__(battery, battery_id)
        self._attr_device_class = SensorDeviceClass.ENERGY
        self._attr_state_class = SensorStateClass.TOTAL_INCREASING
        self._attr_native_unit_of_measurement = UnitOfEnergy.WATT_HOUR
        self._attr_name = f"Sax {battery_id.replace('_', ' ').title()} Energy Produced"

    @property
    def native_value(self):
        """Return the native value of the sensor."""
        hourly_value = self.battery.data.get(SAX_ENERGY_PRODUCED)
        self.update_cumulative_value(hourly_value)  # Update the cumulative total
        return self.cumulative_value

class SAXBatteryEnergyConsumedSensor(SAXBatterySensor):
    """SAX Battery Energy Consumed sensor."""

    def __init__(self, battery, battery_id) -> None:
        """Initialize the sensor."""
        super().__init__(battery, battery_id)
        self._attr_device_class = SensorDeviceClass.ENERGY
        self._attr_state_class = SensorStateClass.TOTAL_INCREASING
        self._attr_native_unit_of_measurement = UnitOfEnergy.WATT_HOUR
        self._attr_name = f"Sax {battery_id.replace('_', ' ').title()} Energy Consumed"

    @property
    def native_value(self):
        """Return the native value of the sensor."""
        hourly_value = self.battery.data.get(SAX_ENERGY_CONSUMED)
        self.update_cumulative_value(hourly_value)  # Update the cumulative total
        return self.cumulative_value
````

Bertk, 2025-03-30:
I think both aspects are part of the issue. The sensor will not update value at the same time when energy dashboard updates the graph.

Bertk, 2025-04-01:
Hi, I used the commit with the fix and also added some sensors to compare results. After a reboot entities "Sax Battery A Energy Produced" and "Sax Battery A Energy Consumed" were set to 0. I think this is related to `SensorStateClass.TOTAL_INCREASING`.

[image]

Sensors for verify results

```
- platform: integration
  name: sax_energy_input
  unique_id: sax_energy_input
  source: sensor.sax_power_input
  method: trapezoidal
  round: 2
  max_sub_interval:
    minutes: 1
- platform: integration
  name: sax_energy_output
  unique_id: sax_energy_output
  source: sensor.sax_power_output
  method: trapezoidal
  round: 2
  max_sub_interval:
    minutes: 1
```

templates for check

```
    # sax-power energy check with intergral helper sensor.sax_battery_combined_power
    - name: sax_power_output
      unique_id: "sax_power_output"
      state_class: measurement
      icon: mdi:mdiBatteryArrowUpOutline
      unit_of_measurement: W
      device_class: power
      state: >-
        {% if is_number(states('sensor.sax_battery_combined_power')) and states('sensor.sax_battery_combined_power') | float(0) > 0 %}
            {{ (states('sensor.sax_battery_combined_power') | float(0)) | round (0) }}
        {% else %}
            {{ float(0) }}
        {% endif %}
      availability: "{{ has_value('sensor.sax_battery_combined_power') }}"
    - name: sax_power_input
      unique_id: "sax_power_input"
      state_class: measurement
      icon: mdi:mdiBatteryArrowDownOutline
      unit_of_measurement: W
      device_class: power
      state: >-
        {% if is_number(states('sensor.sax_battery_combined_power')) and states('sensor.sax_battery_combined_power') | float(0) < 0 %}
            {{ (states('sensor.sax_battery_combined_power') | float(0)) *-1| round (0) }}
        {% else %}
            {{ float(0) }}
        {% endif %}
      availability: "{{ has_value('sensor.sax_battery_combined_power') }}"
```

matfroh, 2025-04-01:
Fix is working on those 2 sensors for me, but it completely changes the way the other sensors are being called and makes my battery pilot useless to steer the battery, the original scope of this component .. so clearly not working…

Now homeassistant wants a total_increasing sensor for the energy dashboard and I want to make that happen. I will probably revert back to the previous sensor, set it as total and create a new one as total increasing, which should add up hourly values and be more precise than any Riemann Sum, as it’s measured by the battery.
Again, it’s important to sum it only once an hour.

Bertk, 2025-04-01:
Looks like energy dashboard works also with `SensorStateClass.TOTAL` and the value is monotonically increasing..

[image]

[image]

matfroh, 2025-04-02:
I reverted back to the previous method, put the class to TOTAL, but that is not how Energy dashboard wants it to be. Will have to go back to the code to make it an increasing value, without affecting the rest of the update methodology.

Here is my sensor for the night. This shows the hourly energy in kwh that was taken from the battery from the sensor and is correct:

[image]

But in the energy dashboard, it does not show the same as it takes the difference from the previous value:

[image]

Am I missing something in the UI? or does the energy dashboard only accept increasing value, to what I found, it's the latter.

matfroh, 2025-04-02:
See my comment above, the energy dashboard does not care about the class, whether total increasing or total, but it wants a cumulative value by the hour, which your trapezoidal integration does.
Now I still want the code to produce such a total_increasing value, but that needs to be calculated by the code in a new sensor. So still work in progress.

Bertk, 2025-04-02:
Correct. :+1:

There is another option to use „TOTAL_INCREASING“ but make sure the monotonically increasing value will not be „0“. I guess this will be sooner or later the wrong way.

https://developers.home-assistant.io/blog/2021/08/16/state_class_total/

matfroh, 2025-04-05:
Adding a new sensor with hourly add-up for the master battery solves it

closed 2025-04-05 by matfroh