matfroh/sax_battery_ha #42: entities are not update after 0.2.2.4 update

Bertk, 2025-06-13:
I noticed some strange values for entity sensor.sax_battery_a_smartmeter after installing the latest update. Only after HA reboot the value is updated for some time.

Issue #40 might a duplicate.

[image]

matfroh, 2025-06-13:
seems related to the python upgrade. 
can you confirm that changing the manifest.json (and restarting) to 
```
"requirements": [
        "pymodbus==3.7.4"
```
solves it?

matfroh, 2025-06-13:
if somebody has some time to test, replace_init_py with below

`"""Integration for SAX Battery - Fixed for pymodbus 3.8+"""

import asyncio
import logging

from pymodbus.client import ModbusTcpClient

from homeassistant.config_entries import ConfigEntry
from homeassistant.const import Platform
from homeassistant.core import HomeAssistant
from homeassistant.exceptions import ConfigEntryNotReady

from .const import (
    CONF_DEVICE_ID,
    CONF_MANUAL_CONTROL,
    CONF_PILOT_FROM_HA,
    DOMAIN,
    SAX_AC_POWER_TOTAL,
    SAX_ACTIVE_POWER_L1,
    SAX_ACTIVE_POWER_L2,
    SAX_ACTIVE_POWER_L3,
    SAX_APPARENT_POWER,
    SAX_CAPACITY,
    SAX_CURRENT_L1,
    SAX_CURRENT_L2,
    SAX_CURRENT_L3,
    SAX_CYCLES,
    SAX_ENERGY_CONSUMED,
    SAX_ENERGY_PRODUCED,
    SAX_GRID_FREQUENCY,
    SAX_PHASE_CURRENTS_SUM,
    SAX_POWER,
    SAX_POWER_FACTOR,
    SAX_REACTIVE_POWER,
    SAX_SMARTMETER,
    SAX_SMARTMETER_CURRENT_L1,
    SAX_SMARTMETER_CURRENT_L2,
    SAX_SMARTMETER_CURRENT_L3,
    SAX_SMARTMETER_TOTAL_POWER,
    SAX_SMARTMETER_VOLTAGE_L1,
    SAX_SMARTMETER_VOLTAGE_L2,
    SAX_SMARTMETER_VOLTAGE_L3,
    SAX_SOC,
    SAX_STATUS,
    SAX_STORAGE_STATUS,
    SAX_TEMP,
    SAX_VOLTAGE_L1,
    SAX_VOLTAGE_L2,
    SAX_VOLTAGE_L3,
)

_LOGGER = logging.getLogger(__name__)

PLATFORMS = [Platform.NUMBER, Platform.SENSOR, Platform.SWITCH]

async def async_setup(hass: HomeAssistant, config):
    """Set up the SAX Battery integration."""
    return True

async def async_setup_entry(hass: HomeAssistant, entry: ConfigEntry):
    """Set up SAX Battery from a config entry."""
    try:
        # Create SAX Battery data instance
        sax_battery_data = SAXBatteryData(hass, entry)
        await sax_battery_data.async_init()
    except (ConnectionError, TimeoutError, ValueError) as err:
        _LOGGER.error("Failed to initialize SAX Battery: %s", err)
        raise ConfigEntryNotReady from err
[... 494 lines cut ...]
                    
                # Parse the bulk response
                for reg_name, reg_info in unit_registers.items():
                    reg_offset = reg_info["address"] - start_address
                    if 0 <= reg_offset < len(result.registers):  # Ensure index is valid
                        raw_value = result.registers[reg_offset]
                        value = (raw_value + reg_info.get("offset", 0)) * reg_info.get("scale", 1)
                        self.data[reg_name] = value
                        # Update the timestamp for this register
                        self._last_updates[reg_name] = current_time
                    else:
                        _LOGGER.warning(
                            "Register %s (address %s) offset %s is out of range for result length %s",
                            reg_name, reg_info["address"], reg_offset, len(result.registers)
                        )
                        
            except (ConnectionError, TimeoutError, ValueError) as err:
                _LOGGER.error("Failed to read bulk registers for unit %s: %s", unit_id, err)
                return False
                
        except TimeoutError as err:
            _LOGGER.error("Error updating battery %s: %s", self.battery_id, err)
            return False
            
        return True`

matfroh, 2025-06-14:
created a hotfix realease to go back to pymodbus 3.7.4. while fixing.

Other extensions with other version requirements of pymodbus are clashing.

reading further pymodbus docs, it seems disconnect are now managed as part of the pymodbus docs:

⚠️ What changed in pymodbus 3.8.x?
Automatic disconnect after 3 failed requests
Now, if your ModbusTCP client makes three requests in a row with no response, pymodbus will automatically close the connection 

New “max until disconnect” setting
Version 3.8.3 added a max_until_disconnect parameter for clients, which controls when long‑lived connections should be torn down 
from: pymodbus.readthedocs.io
.

🧩 Why it works for ~10 minutes, then fails

Opens a persistent TCP connection via ModbusTcpClient(..., timeout=10).

Periodically reads registers—maybe every few seconds.

If three reads in a row don’t get a response, the library closes the socket.

The next read attempt then fails with [Errno 32] Broken pipe, because the TCP socket is already closed.

Bertk, 2025-06-15:
> seems related to the python upgrade. can you confirm that changing the manifest.json (and restarting) to
> 
> ```
> "requirements": [
>         "pymodbus==3.7.4"
> ```
> 
> solves it?

This did not solve the issue. The problem occurs several hours after a HA reboot. I installed latest update **0.2.2.4.1**  and recognized a **typo in manifest.json** ["pymodbu==3.7.4"](https://github.com/matfroh/sax_battery_ha/blob/5b61e5dbc06a9018fde92141bde9c8d7a1f1f37e/custom_components/sax_battery/manifest.json#L15C9-L15C25).

[image]

matfroh, 2025-06-15:
Thanks for finding the typo which prevents the downgrade. Just released again., it still fixes it, but pymodbus needs to be in 3.7.4 and not 3.8.3.
By the way, this manifest.json was the only change since last working release. Make sure you run pymodbus 3.7.4 in this way:
login to the docker container and request pymodbus version:
[image]

Bertk, 2025-06-15:
> if somebody has some time to test, replace_init_py with below
and have requirements set to
"pymodbus>=3.7.4,<3.9.0"

I will check this tomorrow but will use ˋ3.8.0ˋ.

Bertk, 2025-06-17:
I check the the sax_battery modbus error messages in home assistant.log again and it occurs 124 times within 19 hours.
Do you have any idea how this can be reduced?

I plan to disable some entities which I am not using and maybe this should be a feature of the integration. The code already has some commented out section for this.

>The sensors entities are disabled per default, as they contain a lot of data.
>
>You can enable the ones you like in HA UI under "Configuration" -> "Entities" -> click on the filter icon on the right -> Check "Show disabled entities" -> Check the ones you like to enable -> Click "ENABLE SELECTED" at the top -> Confirm the next dialog

| Modbus Error| count|
|:-----|:-----:|
|[Errno 104] Connection reset by peer | 90 |
|No Response received from the remote slave/Unable to decode response| 16 |
|[Invalid Message] No response received, expected at least 8 bytes (0 received)| 9 |
|timed out| 7 |
|[Connection] ModbusTcpClient 192.168.178.90:502| 2 |

```
2025-06-16 13:25:17.059 ERROR (MainThread) [custom_components.sax_battery] Modbus error reading address 47: Modbus Error: [Input/Output] [Errno 104] Connection reset by peer
...
2025-06-16 14:14:25.774 ERROR (MainThread) [custom_components.sax_battery] Modbus error reading address 47: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
...
2025-06-16 14:53:25.809 ERROR (MainThread) [custom_components.sax_battery] Modbus error reading address 47: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
...
2025-06-17 00:52:29.029 ERROR (MainThread) [custom_components.sax_battery] Modbus error reading address 47: Modbus Error: [Input/Output] Modbus Error: [Connection] ModbusTcpClient 192.168.178.90:502
...
2025-06-17 01:14:28.052 ERROR (MainThread) [custom_components.sax_battery] Modbus error reading address 47: Modbus Error: [Input/Output] timed out
...
2025-06-17 08:43:25.608 ERROR (MainThread) [custom_components.sax_battery] Modbus error reading address 47: Modbus Error: [Input/Output] [Errno 104] Connection reset by peer
```

[sax_battery_modbus_errors.zip](https://github.com/user-attachments/files/20771229/sax_battery_modbus_errors.zip)

Bertk, 2025-06-18:
I currently test  pymodbus 3.8.6 update with small changes in `__init__.py`. This did not show errors until now but I will roll back the updates because another integration (heat pump) is broken.

```
homeassistant:/config# pip show pymodbus
Name: pymodbus
Version: 3.8.6
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/pymodbus-dev/pymodbus/
Author: Galen Collins, Jan Iversen
Author-email: 
License: BSD-3-Clause
Location: /usr/local/lib/python3.13/site-packages
Requires: 
Required-by: pyiskra, pysmarty2, pystiebeleltron
homeassistant:/config# 
```

[sax_battery_updates.zip](https://github.com/user-attachments/files/20792248/sax_battery_updates.zip)

Bertk, 2025-06-18:
I tried pymodbus V3.8.6 but this failed. I guess a reconnect after an error occurs might be helpful.

e.g. sample https://github.com/pymodbus-dev/pymodbus/blob/dev/examples/contrib/solar.py
```
        if error:
            error = False
            client.close()
            sleep(0.1)
            client.connect()
            sleep(1)
```
or another integrationdoes it this way  
https://github.com/OStrama/weishaupt_modbus/blob/206ed635152d12ec5de030f0dd4220b046a4e535/custom_components/weishaupt_modbus/modbusobject.py#L47C1-L54C48

```
2025-06-18 10:35:01.291 INFO (MainThread) [homeassistant.setup] Setting up sax_battery
2025-06-18 10:35:01.292 INFO (MainThread) [homeassistant.setup] Setup of domain sax_battery took 0.00 seconds
2025-06-18 10:35:01.295 INFO (MainThread) [custom_components.sax_battery] Successfully connected to battery at 192.168.178.90:502
2025-06-18 10:35:07.622 INFO (MainThread) [homeassistant.components.number] Setting up sax_battery.number
2025-06-18 10:35:07.625 INFO (MainThread) [homeassistant.components.sensor] Setting up sax_battery.sensor
2025-06-18 10:35:07.650 INFO (MainThread) [homeassistant.components.switch] Setting up sax_battery.switch
2025-06-18 10:40:10.207 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.sax_battery_cumulative_energy_going_into_the_battery, value dropped from 5886.0 to 0.0, triggered by state with last_updated set to 2025-06-18T08:35:07.648704+00:00
2025-06-18 10:40:10.207 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.sax_battery_cumulative_energy_coming_out_of_the_battery, value dropped from 7020.0 to 0.0, triggered by state with last_updated set to 2025-06-18T08:35:07.649590+00:00
2025-06-18 10:50:37.692 ERROR (MainThread) [custom_components.sax_battery] General error reading register 47: [Errno 104] Connection reset by peer
2025-06-18 10:50:37.706 ERROR (MainThread) [custom_components.sax_battery] General error reading register 47: [Errno 32] Broken pipe
```

matfroh, 2025-08-17:
> Hi,
> 
> I’m also experiencing issues with the integration after upgrading to Home Assistant 2025.7. Because solaredge_modbus now requires pymodbus>=3.9.2, I updated manifest.json of sax_battery to also use pymodbus==3.9.2.
> 
> However, since then, the integration no longer works properly and throws these warnings and errors:
> 
> ```
> 2025-07-07 11:29:49.495 WARNING (MainThread) [homeassistant.components.sensor] Updating sax_battery sensor took longer than the scheduled update interval 0:00:30
> 2025-07-07 11:29:49.536 WARNING (MainThread) [homeassistant.components.switch] Updating sax_battery switch took longer than the scheduled update interval 0:00:30
> 2025-07-07 11:29:52.097 ERROR (MainThread) [custom_components.sax_battery] Failed to read register 48: [Errno 104] Connection reset by peer
> ```
> 
> I noticed there is work in progress in this fork/branch: 👉 [Bertk#4](https://github.com/Bertk/sax_battery_ha/issues/4)
> 
> Would it perhaps be possible to adapt that branch for pymodbus==3.9.2 and merge it when ready? Or is there another recommended approach for running the integration with the current Home Assistant and pymodbus 3.9.x?
> 
> Thanks a lot for your work and support!

hi, Bertk's branch is working on the next version of HA, which then requires 3.11.1.
I have made progress on 3.9.2 with this [branch](https://github.com/matfroh/sax_battery_ha/tree/upgrade-to-pymodbus-3.9.2), but not yet ready.

matfroh, 2025-08-24:
[pre-release](https://github.com/matfroh/sax_battery_ha/releases/tag/v0.2.3.0beta) ready for testing, looking for tester

closed 2025-08-24 by matfroh