matfroh/sax_battery_ha #47: PV power missing

BA2000, 2025-07-13:
Hi matfroh, 
my setup is a SAX 5.8 with smart Meter ADW200, that can measure the power of the PV also. 
I do Not Control the battery, only monitor. 
On the Web App of SAX i see, SOC, storage power, PV power and grid power.
Is it possible to See the PV power also?

Greetings BA2000

matfroh, 2025-07-13:
Yes, all what is or has been documented in the API by the manufacturer has been added as sensor. Make sure you activate all sensors and then read the value given.
As I do not have this meter, i cannot make sense of the values. I invite you to compare the values and start documenting here what you read on the sensors versus in the webapp. Powerfactor might be uncorrect and might need adjustment.
Also, it might be that the register to read the value was not documented and then I'm unable to guess which register has been used. 

[image]

BA2000, 2025-07-16:
Okay, you can close this issue. I've learned from SAX that this information isn't currently available.
They're working on implementing the SunSpec protocol. This protocol is also capable of delivering PV power.

So we'll just have to wait and see.

btw: thanks for your custom component sax_battery_ha

closed 2025-07-16 by BA2000