matfroh/sax_battery_ha #52: SAX do not start after update to 2025.8.1

BA2000, 2025-08-16:
HI,

it seems that theres an error, SAX do not start anymore.

---
Logger: homeassistant.setup
Quelle: setup.py:278
Erstmals aufgetreten: 14:49:55 (1 Vorkommnis)
Zuletzt protokolliert: 14:49:55

Setup failed for custom integration 'sax_battery': Requirements for sax_battery not found: ['pymodbus==3.7.4'].
---

Greetings BA2000

BA2000, 2025-08-24:
Hi, here is the log:

Logger: homeassistant.setup
Quelle: setup.py:278
Erstmals aufgetreten: 10:23:38 (1 Vorkommnis)
Zuletzt protokolliert: 10:23:38

Setup failed for custom integration 'sax_battery': Requirements for sax_battery not found: ['pymodbus==3.7.4'].

BA2000, 2025-08-24:
Ok, my fault. Because the hint to update came up, but it was a downgrade to the non beta version bevor.
No error occure now.
Here are my 57 entities, for your information i only own one of these sax batteries.

[image]

[image]

[image]

Thanks for your work.

matfroh, 2025-08-24:
thanks for sharing, i believe that i went back to SAX all caps sensor description, which recreates new sensor instead of upgrading the old ones.. will work on that

matfroh, 2025-08-24:
release with major error cleaned live. Now still a few sensor to fix, like power factor and apparent power, storage status, reactive power, mostly naming convention or deactivated for some reasons, need to investigate

matfroh, 2025-08-27:
missing sensors or not properly calculated part of next release.
Looking for more reviewers and code cleaners.