matfroh/sax_battery_ha #63: Unable to set maximum charge power

Bertk, 2025-09-08:
**Describe the bug**
I use an automation which controls charging power which worked in the past but fails after update.

**To Reproduce**
Steps to reproduce the behavior:
1. Use slider for number entity `number.maximum_charge_power` and set a value e.q. 1000 W
2. Check `home-assistant.log` and you will see **pymodbus** errors
3. Check a slider for number entity `number.maximum_charge_power` again and see value is 3500 W

**Expected behavior**
The 1000 W is shown for `number.maximum_charge_power`

**Home assistant server (please complete the following information):**
 - OS:  home assistant yellow V16.1
 - Core: 2025.9.1
 - Front end:  20250903.3

**Hint**
I assume the [SAX Battery has a problems with modbus transaction IDs](https://community.simon42.com/t/einbindung-speicher-ueber-modbus-tcp/13450/120?u=bert42).

Maybe there is a workaround which can be used:

write_register has a parameter no_response_expected which might be helpful. 

> If no response is expected to a request, the no_response_expected=True argument can be used in the normal API calls, this will cause the call to return immediately with ExceptionResponse(0xff).

_I am still working on the SAX battery integration and this will take some more weeks. I decided to use pymodbus synchron client `ModbusTcpClient` and not `AsyncModbusTcpClient`. I want to use charging from grid and hope this can be done before November this year._ 

```python
   def write_registers(
        self,
        address: int,
        values: list[int],
        *,
        device_id: int = 1,
        no_response_expected: bool = False
    ) -> T:
        """Write registers (code 0x10).

        :param address: Start address to write to
        :param values: List of values to write
        :param device_id: (optional) Modbus device ID
        :param no_response_expected: (optional) The client will not expect a response to the request
        :raises ModbusException:

        This function is used to write a block of contiguous registers
        (1 to approx. 120 registers) in a remote device.
        """
```

**Additional context**
_automation fails with error and value is not configured._
 ```text
2025-09-08 09:20:00.460 INFO (MainThread) [homeassistant.components.automation.sax_a_monitor_soc] Sax_a monitor Entladeleistung: Running automation actions
2025-09-08 09:20:00.468 INFO (MainThread) [homeassistant.components.automation.sax_a_monitor_soc] Sax_a monitor Entladeleistung: Choose at step 1: choice 4: Running automation actions
2025-09-08 09:20:00.469 INFO (MainThread) [homeassistant.components.automation.sax_a_monitor_soc] Sax_a monitor Entladeleistung: Choose at step 1: choice 4: Executing step call service
2025-09-08 09:20:00.475 ERROR (MainThread) [pymodbus.logging] ERROR: request ask for transaction_id=1761 but got id=225, Skipping.
>>>>> send: 0x6 0xd8 0x0 0x0 0x0 0x6 0x28 0x3 0x9c 0xa5 0x0 0x1
>>>>> recv: 0x6 0xd8 0x0 0x0 0x0 0x5 0x28 0x3 0x2 0x1 0x9 extra data: 
>>>>> send: 0x6 0xd9 0x0 0x0 0x0 0x6 0x28 0x3 0x9c 0xa6 0x0 0x1
>>>>> recv: 0x6 0xd9 0x0 0x0 0x0 0x5 0x28 0x3 0x2 0x0 0xed extra data: 
>>>>> send: 0x6 0xda 0x0 0x0 0x0 0x6 0x28 0x3 0x9c 0xa7 0x0 0x1
>>>>> recv: 0x6 0xda 0x0 0x0 0x0 0x5 0x28 0x3 0x2 0xff 0x8d extra data: 
>>>>> send: 0x6 0xdb 0x0 0x0 0x0 0x6 0x28 0x3 0x9c 0xa8 0x0 0x1
>>>>> recv: 0x6 0xdb 0x0 0x0 0x0 0x5 0x28 0x3 0x2 0x0 0x38 extra data: 
>>>>> send: 0x6 0xdc 0x0 0x0 0x0 0x6 0x28 0x3 0x9c 0xa9 0x0 0x1
>>>>> recv: 0x6 0xdc 0x0 0x0 0x0 0x5 0x28 0x3 0x2 0x0 0x37 extra data: 
>>>>> send: 0x6 0xdd 0x0 0x0 0x0 0x6 0x28 0x3 0x9c 0xab 0x0 0x1
>>>>> recv: 0x6 0xdd 0x0 0x0 0x0 0x5 0x28 0x3 0x2 0x9 0x41 extra data: 
>>>>> send: 0x6 0xde 0x0 0x0 0x0 0x6 0x28 0x3 0x9c 0xac 0x0 0x1
[... 48 lines cut ...]
>>>>> recv: 0x9b 0x10 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x80 0x0 extra data: 
>>>>> send: 0x9b 0x11 0x0 0x0 0x0 0x6 0x1 0x4 0x89 0x23 0x0 0x1
>>>>> recv: 0x9b 0x11 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0x0 extra data: 
>>>>> send: 0x9b 0x12 0x0 0x0 0x0 0x6 0x1 0x4 0x89 0x24 0x0 0x1
>>>>> recv: 0x9b 0x12 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0x0 extra data: 
>>>>> send: 0x9b 0x13 0x0 0x0 0x0 0x6 0x1 0x3 0xb0 0x2d 0x0 0x1
>>>>> recv: 0x9b 0x13 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x1 extra data: 
>>>>> send: 0x9b 0x14 0x0 0x0 0x0 0x6 0x1 0x3 0xb0 0x2e 0x0 0x1
>>>>> recv: 0x9b 0x14 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2 extra data: 
>>>>> send: 0x9b 0x15 0x0 0x0 0x0 0x6 0x1 0x3 0xb0 0x2f 0x0 0x1
>>>>> recv: 0x9b 0x15 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x3 extra data: 
>>>>> send: 0x9b 0x16 0x0 0x0 0x0 0x6 0x1 0x3 0xb0 0x30 0x0 0x1
>>>>> recv: 0x9b 0x16 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x4 extra data: 
>>>>> send: 0x9b 0x17 0x0 0x0 0x0 0x6 0x1 0x3 0xb0 0x33 0x0 0x1
>>>>> recv: 0x9b 0x17 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0xff 0xff extra data: 
>>>>> send: 0x9b 0x18 0x0 0x0 0x0 0x6 0x1 0x3 0xb0 0x34 0x0 0x1
>>>>> recv: 0x9b 0x18 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0xff 0xff extra data: 
>>>>> send: 0xc 0xa3 0x0 0x0 0x0 0x9 0x40 0x10 0x0 0x2c 0x0 0x1 0x2 0x4 0xb0
>>>>> recv: 0x0 0xa3 0x0 0x0 0x0 0x6 0x40 0x10 0x0 0x2c 0x0 0x1 extra data: 
2025-09-08 09:31:52.342 ERROR (MainThread) [pymodbus.logging] ERROR: request ask for transaction_id=3235 but got id=163, Skipping.
>>>>> send: 0xc 0xa3 0x0 0x0 0x0 0x9 0x40 0x10 0x0 0x2c 0x0 0x1 0x2 0x4 0xb0
>>>>> recv: 0x0 0xa3 0x0 0x0 0x0 0x6 0x40 0x10 0x0 0x2c 0x0 0x1 extra data: 
2025-09-08 09:31:57.348 ERROR (MainThread) [pymodbus.logging] No response received after 1 retries, continue with next request
2025-09-08 09:31:57.348 ERROR (MainThread) [custom_components.sax_battery.number] Failed to write max charge value: Modbus Error: [Input/Output] No response received after 1 retries, continue with next request
```

matfroh, 2025-09-08:
thanks for spotting, missed that one. In the previous versions I simply ignored the response. I'll have to see why it was lost. Seems the max values are also wrong.

Bertk, 2025-09-10:
@matfroh I modified my implementation of SAX battery write_registers for limit number entities. 
See commit [use no_response_expected for WO registers](
https://github.com/Bertk/sax_battery_ha/commit/766e4ce5c9600f0d968fd9b6a41d1a81214dd324)

I do not have a development environment for the current main branch and did not verify the power status after changing the limits.